tube0013/TubesZB-ESPHome-Builder #42: [ESPHome] New Configuration Request

ecemaster, 2025-02-09:
### Device_Name

tubeszb-zw

### Static_IP

192.168.1.34

### Gateway

192.168.1.1

### Subnet

255.255.255.0

ecemaster, 2025-02-09:
....
Creating esp32 image...
Successfully created esp32 image.
Linking .pioenvs/tubeszb-zw/firmware.elf
RAM:   [          ]   4.3% (used 14252 bytes from 327680 bytes)
Flash: [===       ]  27.2% (used 498513 bytes from 1835008 bytes)
Building .pioenvs/tubeszb-zw/firmware.bin
Creating esp32 image...
Successfully created esp32 image.
esp32_create_combined_bin([".pioenvs/tubeszb-zw/firmware.bin"], [".pioenvs/tubeszb-zw/firmware.elf"])
Wrote 0x89cc0 bytes to file /data/build/tubeszb-zw/.pioenvs/tubeszb-zw/firmware.factory.bin, ready to flash to offset 0x0
esp32_copy_ota_bin([".pioenvs/tubeszb-zw/firmware.bin"], [".pioenvs/tubeszb-zw/firmware.elf"])
========================= [SUCCESS] Took 75.61 seconds =========================
INFO Successfully compiled program.
INFO Connecting to 192.168.1.34 port 3232...
INFO Connected to 192.168.1.34
INFO Uploading /data/build/tubeszb-zw/.pioenvs/tubeszb-zw/firmware.bin (498880 bytes)
ERROR Error binary size: Unknown error from ESP

ecemaster, 2025-02-09:
INFO ESPHome 2024.12.4
INFO Reading configuration /config/esphome/tubeszb-zw.yaml...
WARNING GPIO12 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
WARNING GPIO5 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Starting log output from 192.168.1.34 using esphome API
INFO Successfully connected to tubeszb-zw @ 192.168.1.34 in 0.001s
WARNING tubeszb-zw @ 192.168.1.34: Connection error occurred: tubeszb-zw @ 192.168.1.34: The connection dropped immediately after encrypted hello; Try enabling encryption on the device or turning off encryption on the client (ESPHome Logs 2024.12.4).
WARNING Can't connect to ESPHome API for tubeszb-zw @ 192.168.1.34: tubeszb-zw @ 192.168.1.34: The connection dropped immediately after encrypted hello; Try enabling encryption on the device or turning off encryption on the client (ESPHome Logs 2024.12.4). (HandshakeAPIError)
INFO Trying to connect to tubeszb-zw @ 192.168.1.34 in the background
INFO Successfully connected to tubeszb-zw @ 192.168.1.34 in 0.002s
WARNING tubeszb-zw @ 192.168.1.34: Connection error occurred: tubeszb-zw @ 192.168.1.34: The connection dropped immediately after encrypted hello; Try enabling encryption on the device or turning off encryption on the client (ESPHome Logs 2024.12.4).

ecemaster, 2025-02-09:
tubeszb-zw GUI:

16:28:07 | [I] | [safe_mode:041] | Boot seems successful; resetting boot loop counter
-- | -- | -- | --
16:28:07 | [D] | [esp32.preferences:114] | Saving 1 preferences to flash...
16:28:07 | [D] | [esp32.preferences:142] | Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
16:28:29 | [D] | [api:103] | Accepted 192.168.1.102
16:28:29 | [W] | [api.connection:107] | : Reading failed: BAD_INDICATOR errno=11
16:28:31 | [D] | [api:103] | Accepted 192.168.1.102
16:28:31 | [W] | [api.connection:107] | : Reading failed: BAD_INDICATOR errno=11

tube0013, 2025-02-09:
looks like a mismatched encryption key possibly with one stored in HA. I don't use encryption in the FW, this only happens if you previously compiled a version yourself I believe.

ecemaster, 2025-02-10:
Couldn't figure out how to turn the encryption off within ESPHome. Did a "Install -> Manual download" of the '.bin' file generated by ESPHome, then did "Install -> Plug into this computer" and flashed that '.bin' file into the TubeZB-ZW via USB Micro on my computer. Works perfect now and can switch my Z-Wave plug on/off in Home Assistant using Z-Wave JW UI.

[image]

closed 2025-02-18 by tube0013